calyxir/calyx #2366: No More Global Symbol Table

ethanuppal, 2024-12-02:
Au revoir
Adiós
Dasvidaniya
Zàijiàn
Arrivederci
Sayonara
Auf Wiedersehen
Adeus
Annyeonghi gasipshio
Iilaa alliqa
Farvel
Kwaheri
Dag
Antio
Do widzenia
Sampai jumpa
Namaste
Adjø
Hoşçakalın
Lehitra’ot
Hej då

We should consider refactoring the code to avoid a global static mutable variable (which is very likely and does currently cause UB). #2365 should be a temporary fix.

EclecticGriffin, 2024-12-02:
To elaborate, the current design has been exhibiting UB for a while, we just got lucky that it never triggered anything bad. It is also fundamentally thread-unsafe which isn't ideal. If we wanted to keep the global position table, we would need to make the `SINGLETON` only ever return an immutable reference and all the mutable methods would have to use interior mutability with some locks. You could then have the read methods return a lock guard instead of a direct reference, or if we're very careful and make it impossible to mutate things after they are added to the table, we could use some unsafe to extend the lifetime (see `symbol_table`'s implementation) and unlink it from the lock to return a reference.

EclecticGriffin, 2024-12-02:
Of course this won't solve the problem of deadlocks which remain something of an issue with the current usage of the symbol table since we often have longstanding borrows with intervening writes, though this could be mitigated with the aforementioned lifetime extensions & unlinking trickery.

EclecticGriffin, 2024-12-03:
Actually I was thinking about this further, and the trick used by `symbol_table` might not actually be safe. I'll need to look into how they do it, but I think there's a risk of dangling pointers when things resize in vecs

EclecticGriffin, 2024-12-03:
Ah yes, okay so their vecs are actually storing boxes of the data instead of storing it directly

ethanuppal, 2024-12-04:
Cloning introduced (lines and files with respect to the result of #2365) that we need to fix:
- `parser.rs`: 
   - `gpos_ref.get_source(file).to_owned()` on 63 and 102 
- `position.rs`: 
   - `fn get_lines`: lines associated with span couldn't be `&'static str`s anymore
   - `fn get_line_num`: has to return an owned `String`
   - `fn get_location`: has to return an owned `String`
   - (not cloning-related) `lines[0].as_str()` is probably ok to be `&lines[0]`

ethanuppal, 2024-12-04:
Would it be a good idea to just get rid of the global state entirely and pass a `&mut PositionTable` or `RRC<PositionTable>` down to the places that need it? If so, I'll start working on that refactor.

ethanuppal, 2024-12-04:
I don't know if doing another hack to make the mutable reference work is a good idea because it already revealed some UB in the existing implementation.

EclecticGriffin, 2024-12-04:
So while I agree that global state isn't ideal, I think overhauling this to pass the Position Table in will actually be quite a lot of work.

The first and more solvable problem is that the current usage seems to intermingle brief mutable accesses with longer ongoing immutable ones. Judicious use of `std::mem::take` might be able to get around this though some care would be needed to not break the index ordering.

The more annoying problem is tracking down every use of `GPosIdx` and trying to weave in references to the table for its methods. This might not be possible in all cases, such as in display and error printing, and is liable to be a serious pain.

Unfortunately, just making a less UB-riddled version of the global table might be the more pragmatic option here. But this is, at least partially, speculation on my part.

EclecticGriffin, 2024-12-12:
I'm sure such a thing is possible, but I don't think it's really a high priority and consequently not a great use of limited development time

closed 2024-12-17 by EclecticGriffin