calyxir/calyx #2427: [CI] Extension deployment token expired

EclecticGriffin, 2025-03-06:
The Extension Deployment action has recently stopped working since the VSCode marketplace access token expired. I don't have access to the Azure devops org for Calyx. I think @sgpthomas might be the only one who does?

sgpthomas, 2025-03-10:
Made a new token, added it, and re-enabled the workflow. Should be good for another year :) But I can also add people to the Azure organization. If you add your user id to this thread, I'll add you to the org. Honestly don't really remember what the "user id" would mean here? I think you might have to make an Azure Devops account? This is probably relevant: https://code.visualstudio.com/api/working-with-extensions/publishing-extension

closed 2025-03-10 by sgpthomas

EclecticGriffin, 2025-03-10:
My ID is apparently: 7f5e843b-4060-698a-8648-e35294dbc9e5

EclecticGriffin, 2025-03-10:
@rachitnigam If you go to this page and sign in: https://marketplace.visualstudio.com/vscode

there will be an ID next to your name that you can copy

EclecticGriffin, 2025-03-19:
@sgpthomas Just wanted to follow up on this!

sgpthomas, 2025-03-23:
Sorry for the delay on this. I've added @EclecticGriffin to the vscode marketplace publisher as an owner, and have sent you an invite to the Azure dev-ops org (not sure if that's necessary? but you might need it to make personal tokens?). I honestly find this whole process rather baffling and am not certain on how the Azure dev-ops is related to the Vscode publisher stuff, but maybe you have more permissions now?